mat1jaczyyy/lpp-performance-cfw #8: Lights not showing

TheLifeGamingYT, 2018-09-21:
Hello, I'm back again... with another problem (probably dumb but still tried everything).
Before the lights did start to not show, I was playing a normal launchpad project. After I tried a dumb thing that I knew that it wouldn't work but I wanted to try. So I launched up a Max patch that would let me use the mode light, but (I knew it!) it didn't work. So I closed Max and I re-tried to reset the port for the output of midi lights. I tried pressing a button... it played the sound but not the lights! So I tried upgrading the firmware but nothing. I closed entirely Max but it didn't work, I tried closing and re-opening Live but it didn't work. I don't have more possible solutions left. Help.

mat1jaczyyy, 2018-09-21:
I'm not sure what exactly it is you're trying to achieve? Do you just want to fix the issue with your light effect feedback, or are you trying to enable mode light feedback?

TheLifeGamingYT, 2018-09-22:
I want to fix the light feedback

mat1jaczyyy, 2018-09-22:
No one can help you if you don't provide any specific information on what you're doing. Will put this on hold until you can detail exactly what it is you have set up.

TheLifeGamingYT, 2018-09-22:
I don't understand what you're saying with "provide specific information". I already said everything I know

mat1jaczyyy, 2018-09-22:
What mode are you using on the Launchpad? How are Ableton's MIDI preferences configured? How is your track output configured?

TheLifeGamingYT, 2018-09-22:
I'm using performance mode, MIDI configuration: Control Surface: Launchpad Pro; Input port: Launchpad Open; Output port: Launchpad Open (Port 2)

TheLifeGamingYT, 2018-09-22:
Oh and track output Launchpad Open (Port 2)

mat1jaczyyy, 2018-09-22:
Track Output should be just "Launchpad Open". Refer to Live setup in README.md

closed 2018-09-22 by mat1jaczyyy

TheLifeGamingYT, 2018-09-22:
But if it's like this why before was working