mat1jaczyyy/lpp-performance-cfw #9: Add support for mode light

TheLifeGamingYT, 2018-09-24:
Hello, I'm back again (I'm sorry mat if I'm stressing you)!
I wanted to ask you if you could add support for the pro mode light in your CFW.
But only support for mode light but even add the feature to listen to the solo light and synchronize it to the mode light like shown in this video: https://www.youtube.com/watch?v=2-BxZaxdY8Y
Probably this could be a big feature to add as for right now we need external max patches but as this CFW doesn't support max then if you add it, will be a game-changer!
Thanks bro

Niekxd, 2018-09-24:
fam you can already do that iirc

TheLifeGamingYT, 2018-09-24:
How?

mat1jaczyyy, 2018-09-24:
Mode light is accessible in Performance mode on D#0 in Drum Rack layout and on 99 in XY layout. If you want to synchronize it to any of the circle or square buttons, create a MIDI Effect Rack that does so.

closed 2018-09-24 by mat1jaczyyy

TheLifeGamingYT, 2018-09-25:
Sorry I didn't know that

TheLifeGamingYT, 2018-09-28:
Ok but how do I create it

TheLifeGamingYT, 2018-09-28:
(sorry I don't know much about midi effects in Ableton)

mat1jaczyyy, 2018-09-28:
that falls outside the scope of this issue.