mat1jaczyyy/lpp-performance-cfw #10: Piano mode

Niekxd, 2018-09-29:
AKAI recently released their FL Studio Fire device, and it has a really cool feature that it lays out the buttons like a piano in this way:
[image]
The green keys are the C keys,
The purple keys are the white keys,
The orange keys are the black keys.

It would be really nice to see something like this implemented on the firmware so that music producers can create melodies and chords way more easier without buying an actual keyboard.
Cheers.

mat1jaczyyy, 2018-09-29:
In the context of the Launchpad, do you think it would be better to have 8x2 panels for octaves? You will for sure need octave selector, but do you think it requires a transposition one (think left/right in note/scale mode)?

Niekxd, 2018-09-29:
8x2 would be fine i guess, the transposition one is very interesting and i really like that idea.

mat1jaczyyy, 2018-09-29:
This will almost certainly take a bit of time to implement. I will take a deeper look when I find some time to focus on it

commit 91f005e1 (2018-10-03): Merge pull request #13 from mat1jaczyyy/piano

Implements piano mode, resolves #10.
PR #13 (2018-10-03): Piano mode
Implements an additional piano-like mode. Resolves #10.
closed 2018-10-03 by mat1jaczyyy