mat1jaczyyy/lpp-performance-cfw #19: Are you going to update this firmware to new version?

DelicatedSauce, 2018-12-31:
Because the light seems delaying than novation firmware.

I've tested rainbow ligthshow in 2 case.
The lightshow was more blinking in this firmware. even i was using ASIO4ALL.

is this built lower than 182 version?

mat1jaczyyy, 2018-12-31:
The custom launchpad firmware is built on the latest and greatest, probably never than 182 internally.

But, the reason for the lags is the inability to use the Novation Driver. Novation is aware, but as usual are taking a while to fix. In the meantime you can attempt to edit the driver's INF file to force it to load for our open source Launchpads. This approach works and fixes our lags. I have a guide pinned in the #dev channel on my discord server for this, but you should only do it if you know what you're doing.

That being said, this does need an official fix, but it is sadly outside my control.

mat1jaczyyy, 2019-05-07:
this has been fixed by #20 in 493df1ca7ae63ad83383e9528e17d4e442231cc7.

closed 2019-05-07 by mat1jaczyyy