mat1jaczyyy/lpp-performance-cfw #29: Multi reset issue

BattaNico, 2019-10-06:
Pressing the "setup" button on LP PRO doesn't reset the Multi device anymore

TheLifeGamingYT, 2019-10-06:
I have this problem too

mat1jaczyyy, 2019-10-06:
Is this in Apollo or Ableton, and what version? Are you using the latest outbreak devices?

TheLifeGamingYT, 2019-10-06:
It's in Ableton and I'm using the latest outbreak

BattaNico, 2019-10-06:
I'm using all the latest version of both outbreak and cfw

mat1jaczyyy, 2019-10-16:
Just found some time to test. Testing with latest CFW, Multi Reset works as intended. Note that the reset trigger happens when you ENTER performance mode, so a transition from setup mode to performance mode should reset your multisamples. Live mode does NOT support this feature. If you're still sure that there's an issue, please re-open and provide a Live set / project file which can consistently reproduce the issue.

closed 2019-10-16 by mat1jaczyyy