mat1jaczyyy/lpp-performance-cfw #30: [Question] Adding a new mode/layout?

lukaszkorecki, 2020-05-02:
Hi! I just need some guidance on how to add a new mode/layout. Not sure how to call this - it would be an equivalent of the custom modes, which are supported by latest Launchpads (Mk3, Mini and X) and can be created via Novation's Components app.

My use case is pretty simple :-) - I'd like to add a custom layout to make the Launchpad to work as a drum pad controller for the MPC One (or Live or X, and possibly any other MPC). In the "drum program" mode - you'd assume that MPC's  16 pads are laid out like this:

```
[C2 ] [C2#] [D2 ] [D2#]
[G1#] [A1 ] [A1#] [B1 ]
[E1 ] [F1 ] [F1#] [G1 ]
[C1 ] [C1#] [D1 ] [D1#]
```

If that was the case - using chromatic layout would work out of the box. Turns out - it's all over the place, [this is the best resource I found that shows the notes <-> pads mapping](https://www.noterepeat.com/products/akai-professional/mpc-series/mpc-renaissance-and-studio/199-akai-mpc-renaissance-and-studio-pad-bank-performance-with-external-controllers). 

The solution would be to add a new layout to LPP, which when activated show 4 4x4 grids ([something like this](https://share.getcloudapp.com/DOuQBK5J)) which would allow for triggering all pads, across all 4 banks from the launchpad.

Hopefully you get the idea.

I don't mind hacking on this - just needs some pointers where to get started, as I'd like to keep other modes (scale mode) and Live integration intact. Thanks!

mat1jaczyyy, 2020-05-02:
If I'm getting this right, you need Drum mode to trigger different notes? The layout shouldn't be too hard to do, but I'm confused on how you want to switch between the layouts for all 8 differently mapped banks (since they can't fit all at once)

lukaszkorecki, 2020-05-02:
@mat1jaczyyy yeah kinda, it would be like a custom drum rack mode, but sending plain old midi-notes. To clarify: it has nothing to do with Ableton Live.

As for the banks - I guess you can only see/use 4 at a time (4 x 16 pads) at a time. To use the other ones you could potentially use the up/down buttons on the LPP. In practice, having access to only first 4 banks is fine, I think.

mat1jaczyyy, 2020-05-03:
@lukaszkorecki Yeah, shouldn't be too hard. Mind forking the repo? I'd push changes to your fork as this is a pretty specific use case

lukaszkorecki, 2020-05-03:
@mat1jaczyyy For sure, I think having my own fork makes way more sense.

So circling back to the original question - what would be a good jumping off point to see how this could be done?

mat1jaczyyy, 2020-05-03:
@lukaszkorecki I made a PR into your fork adding MPC mode. Hopefully you can finish it up and learn how adding a mode works!

lukaszkorecki, 2020-05-05:
@mat1jaczyyy woah, awesome! Thank you - I'll check it out!

closed 2020-06-01 by mat1jaczyyy