mat1jaczyyy/lpp-performance-cfw #44: (Chord Mode) Saved chords record other saved chords by pressing both at the same time.

rasch3n, 2021-02-27:
With the MK3 you can play chords simultaneously while in record mode without them recording each other and thus ruining themselves.

mat1jaczyyy, 2021-02-27:
There's no way for my system to differentiate whether an input came from the grid or from the chord bank... I'll see what I can do so it doesn't ruin other saved chords at least

rasch3n, 2021-02-27:
Here is how it should work 
https://youtu.be/4iNJ0Om5ctg

rasch3n, 2021-08-15:
Hi mat
Is there anything new on this subject? :)

jaard, 2021-09-01:
It's still happening I think. What would be needed to differentiate the "chord bank" button presses from the other pads? Some sort of intermediate layer? I'm interested in contributing if I can, but I would need some pointers. I also have some ideas for a slightly different chord mode that I want to work on, but this issue may be a good way to get familiar with the code first.

mat1jaczyyy, 2021-09-14:
I've been missing all my GitHub notifications lmao, sorry. I'll get around to these eventually, but no guarantee on when.