mat1jaczyyy/lpp-performance-cfw #45: LPX support

awesome-coder-d, 2021-02-28:
Hello mat1jaczyyy,

I think there should be a CFW for Launchpad X. I own one, so I think it would be helpful.

conc1erge, 2021-02-28:
Not gonna speak for Mat here but:

Novation didn't release an open source version of the X's firmware, Only the Pro MK1.
Without the full source of the X's firmware, a lot of reverse engineering has to be done and nobody has time for that.

RagnaLP, 2021-02-28:
lpx fw isnt open source, so its either gonna be reverse engineered to shit or it isnt gonna exist, everything you need is on the stock firmware anyway

SanJacobs, 2021-03-04:
Since I don't see any information about this in the readme, and this seems somewhat related, is this compatible with the Launchpad Pro MK3?

RagnaLP, 2021-03-04:
no, each lp has different fw even if it performs similarly

awesome-coder-d, 2021-03-04:
thanks for telling me conc1erge. I am typing this reply via email. LOL

mat1jaczyyy, 2021-03-07:
Interfacing with hardware is obviously different if you have different hardware units. Any naming scheme similarity can't mean the hardware is remotely similar, even if looks the same on the surface.

Novation released libraries for talking to the LP Pro hardware only, all other models don't have this. While we can manually figure out how to talk to the hardware by reversing stock firmwares, and this is something that's near impossible to do (but still technically can be done).

closed 2021-03-07 by mat1jaczyyy

SanJacobs, 2021-03-08:
I'd suggest writing that this is only for the mark 1 in the readme. I doubt I'm the only person who has been slightly confused by the use of just "Pro", since there are multiple Pro models.