mat1jaczyyy/lpp-performance-cfw #50: (Custom Modes) Toggled MIDI notes do not indicate status

tsmorrill, 2021-03-14:
MIDI notes that are set to toggle do not light up when pressed.

mat1jaczyyy, 2021-03-14:
This worked on my build, can you please send a custom mode that contains faulty behavior?

rasch3n, 2021-03-15:
I have some same kind of issue but just if I have 2 custom pages witch toggles. One is working and the other one not.

tsmorrill, 2021-03-15:
Here is my custom mode. https://www.dropbox.com/s/y2qgxjjio2c7xwd/Harmonic%20Hold.syx?dl=0 I had this in the second custom mode slot.

rasch3n, 2021-03-15:
in my setup also just the first page was working!

mat1jaczyyy, 2021-03-15:
Thanks. My Launchpad Pro is out of commission at the moment, I'll check as soon as I'm back to a working state.

rpgaleksy, 2021-03-17:
This is intended behavior.
If velocity sensitivity is enabled you need to press a button at 127 (maximum) velocity for the toggles to switch state.
Disabling velocity sensitivity results in the behavior you expect.
This is exactly how the Launchpad X does it as well.

tsmorrill, 2021-03-20:
Clarification, is it intended behavior to have a mismatch between visual state and MIDI state? As is, I can have a note held at 96 velocity, but the interface will not reflect that the note is active.

rpgaleksy, 2021-03-20:
Is that so? I'll check that out and compare CFW to X firmware shortly,
As far as I know, mat1 recreated the Xs behavior 1-to-1 in CFW custom modes.

mat1jaczyyy, 2021-03-22:
> 
> 
> Clarification, is it intended behavior to have a mismatch between visual state and MIDI state? As is, I can have a note held at 96 velocity, but the interface will not reflect that the note is active.

@tsmorrill This issue exists on the Launchpad X and Mini MK3 when using custom modes "the official way" in a velocity sensitive environment. Whether Novation intended this behavior in their official implementation is up for debate.

When talking about CFW though, it is considered intended behavior because our goal is to implement the custom modes as accurately as possible to the Launchpad X, making sure there are as little differences as possible.

mat1jaczyyy, 2021-04-04:
While I understand it's confusing, I'm closing as this is intended behavior

closed 2021-04-04 by mat1jaczyyy