mat1jaczyyy/lpp-performance-cfw #52: [Custom mode] program change "on color" not working

benderskiy, 2021-03-26:
Using it with an external midi, I have 8 programs in column per channel and switching works like a charm, however color of activated pad never changes to the one I've seletec as "on color" in a custom mode builder. I've tried the same mode before with LPP mk3 and the same has worked, so might be bug?
P.S. huge thanks to @mat1jaczyyy and the community for such a great firmware!

mat1jaczyyy, 2021-04-04:
@benderskiy hey there, sorry for the late reply. could you attach the custom mode that displays the problem here, in a zip if possible? thank you!

benderskiy, 2021-04-06:
an offtopic question (tell me if I should open up a new thread if not appropriate): is it possilbe to define / change CC messages sent by side buttons? is there e.g. a custom config for it? with my mc-707 "shift" is changing cutoff of the selected track, which is a bit annoying :-)

risseraka, 2021-05-07:
Hi @benderskiy.
I had a look at this and proposed a fix.
As a side note, you might want to chose another color than red for your on value as your first column is already red and you wouldn't see any change.
Cheers.

mat1jaczyyy, 2021-05-08:
hey @benderskiy, i've finally had a look at this.

on my Launchpad X, the program changes emit no on color, just like on the current CFW implementation. since we want to stay as close as possible to the LPX, we won't be implementing this even if the LPP MK3 handles on color for this case.

if you absolutely need this functionality, feel free to use a custom build, @risseraka's code might work out for you.

> is it possilbe to define / change CC messages sent by side buttons? is there e.g. a custom config for it? with my mc-707 "shift" is changing cutoff of the selected track, which is a bit annoying :-)

not without recompiling, you can change that behavior here: https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/8ef1e143dd7198390ecd9e9b8b4ae06afe0cfa50/src/modes/normal/custom.c#L567

cheers

risseraka, 2021-05-09:
@benderskiy link to firmware with aforementioned fix-which-isn't-really-a-fix. 😄 
Cheers.

https://github.com/risseraka/lpp-performance-cfw/raw/a92ac4cb0b582b463d63646566902287fef3038a/build/cfw.syx

benderskiy, 2021-05-11:
@risseraka fire! You’re awesome, thnx, man!

closed 2021-05-11 by benderskiy