mat1jaczyyy/lpp-performance-cfw #56: LED does not flash between last set color and given color

risseraka, 2021-05-04:
Hi mat1jaczyyy,

First and foremost, you're doing a fantastic job.

It seems that flashing is note quite the same with the original firmware.

Originally, flashing is done between the last set color and the given color.

Example:
- set pad to 64 (green)
- set flashing to 5 (red)
- -> expected: LED flashes between green and red
- -> actual: LED flashes between red and black

Original behavior is very helpful and can be used to create alternating patterns depending on last set color (black to red or red to black) along with internal tempo.

As that's what I'm using to develop a sequencer, I've only tested this in programmer mode.

(Besides this issue, I'm wondering if you could implement RGB grid flashing/pulsing too, instead of 0-127 color codes, that would be a huge help! Let me know if I should create another separate issue.)

Thanks for your work.

Cheers,
Adrien.

mat1jaczyyy, 2021-05-05:
Hey there,

I vaguely recall this behavior being a thing at some point. I'll definitely explore what's going on once I find the time, I'm super busy these days...

RGB flashing / pulsing is implemented for Performance mode's custom palettes, but isn't directly accessible yet. We can certainly make a SysEx API for it, whether it's a regular one or Apollo-compression like. Do create a separate issue for that so I can understand your needs better 👍

risseraka, 2021-05-05:
Hey,

Thanks for coming back to me!

Idea for fixing this behavior, adding having "led_screen" array to be able to flash back and forth between previously set led color and flashing color.

https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/7aaf6445a8ffe2db15898efb2e5806d70d404816/src/led/led.c#L3

I've created another issue for discussion about sysex implementation of RGB pulsing/flashing.
https://github.com/mat1jaczyyy/lpp-performance-cfw/issues/57

Again, impressive work.

As said in the other issue, I might try to do these myself, but C programming skills are lacking.
Will keep you posted.

Cheers.

commit 74bcc9f3 (2021-05-07): LED: Fix flashing back and forth between previous set color and flash color, closes #56
closed 2021-05-07 by mat1jaczyyy