mat1jaczyyy/lpp-performance-cfw #57: Make RGB pulsing / flashing available through Sysex

risseraka, 2021-05-05:
Hi,

It would be great if `flash_rgb_led` & `pulse_rgb_led` were accessible using Sysex.

I'm developing a JS sequencer and using the Launchpad Pro as a display and controller.
Representing different notes or patterns groups, I chose to use different basic colors (green, blue, orange, cyan, etc.).

Using RGB would enable me to flash or pulse a brightness gradient, which isn't directly feasible using color codes at the moment as it goes only to 127 colors codes.
(Side note, I spent two days making code correspond to RGB one by one, before realizing you had them listed in your project as a palette, oops, ha ha).

Example:
If I wanted to have 12 different values for green, starting from 3, I would like to be able to flash or pulse with these values:
[0, 3, 0],
[0, 9, 0],
[0, 15, 0],
[0, 21, 0],
[0, 27, 0],
[0, 33, 0],
[0, 39, 0],
[0, 45, 0],
[0, 51, 0],
[0, 57, 0],
[0, 3, 0],
[0, 63, 0]

It should be farely easy to implement, but my C programming dates back from 15 years ago, so I'm not too confident, I'll try.
Here's where I guess it should go: https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/master/src/sysex/sysex.c#L220

Cheers.

mat1jaczyyy, 2021-05-05:
Yeah, should go somewhere around there. My SysEx stuff is macro-heavy in particular, but adding this shouldn't be too hard. I'd just prefer if it didn't use Novation's official header, but rather just a simple small and fast message. It should be done similarly to this regular RGB set:

https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/7aaf6445a8ffe2db15898efb2e5806d70d404816/src/sysex/sysex.c#L104-L108

Also, assuming you're using Programmer mode, you'll most definitely be losing access to `programmer_screen` this way, so entering setup and exiting back won't restore whatever the LED state was previously. Honestly I'd prefer that to stay that way as keeping yet another buffer of these is going to take up more memory than I'd like...

Because you're not using the screen, those won't redraw when the user goes to setup and back, so to deal with that you'll most likely write your app to completely use SysEx RGB sets so you don't use any of the screens, at least avoiding a partial state. I guess then you'd just need some way to tell when the user has re-entered the mode to redraw stuff.

This is kinda why flashing went extremely underused even dating back to the original non-RGB Launchpad models, it's just too gimmicky to work with. No one wants to send a MIDI tempo clock unless they really have to, and there are many edge cases to think about where things can go haywire. Do also consider completely driving the display in JS, without expecting the firmware to do the flashing for you (simulate the flashing in JS itself).

risseraka, 2021-05-05:
I had this issue on Windows after installing everything and executing `make`.
```
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/cfw.elf section `.bss' will not fit in region `RAM'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 46980 bytes
collect2.exe: error: ld returned 1 exit status
```

Then I tried on my Linux computer and got this one at the end of `make`
```
./tools/bintosyx /pro 000 build/cfw.bin build/cfw.syx
./tools/bintosyx: ./tools/bintosyx: cannot execute binary file
```

So, I copied both `elf` and `bin` files back on Windows and built the `syx` file there, it worked!

---

Anyway, I didn't fully get what you meant in your comment, being unfamiliar with how modes work.
I went ahead and implemented this issue in my custom firmware in #58, works like a charm.

Another PR will arrive shortly, but as I did like you told me not to, it will only be up for discussion. 😁 

---

Re "driving the display in JS": been there, done that, keeping track of the timeouts while changing screens (e.g. step sequencer to arrangement view) is a lot of pain, and I had a lot of flashing colors because everything wasn't cancelled correctly.
When I found out about flash/pulsing in the programmer guide, it felt liberating as I wouldn't have to handle all that, but no RGB support was underwhelming.

Shall we discuss this topic in #56? (I.e. flashing between two colors.)

Cheers.

risseraka, 2021-05-05:
Looking at this line, I might have understood a bit more what you said about changing modes.
For flashing to work with two colors, you would have to save the state of the whole colors/flashing pads on the programmer mode.
https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/master/src/modes/normal/programmer.c#L53

I think I could live with that, I don't plan on switching from programmer mode to, say, piano mode.
In fact, I've already re-iplemented piano mode to add a few functionalities, like scale transposition, chords mode on/off, etc.

I'll take some more time looking at your comment and digest it.

In the meantime, I'll set my PR to draft and see later.

Cheers.

mat1jaczyyy, 2021-05-05:
> region `RAM' overflowed by 46980 bytes
Not on my machine, idk

> Then I tried on my Linux computer
The bintosyx executable there is intended for macOS.

commit 354d05eb (2021-05-05): Sysex: add RGB fast flashing and pulsing, closes #57
closed 2021-05-05 by mat1jaczyyy