mat1jaczyyy/lpp-performance-cfw #62: Transpose in Custom Modes

mattedwards2686, 2021-06-21:
Hello,
Would it be possible to enable the transpose buttons in Custom Modes?  I use Custom Modes to control different instruments, and it would be very helpful to have the transpose functionality work, or to have the transpose buttons send a CC.

Thank you for all of your work!

mat1jaczyyy, 2021-06-21:
I'm assuming you mean the Up/Down/Left/Right buttons with "transpose buttons". They do send CCs on the latest edition of the firmware. I'm not entirely sure what you're asking for here?

mattedwards2686, 2021-06-21:
My apologies, I didn't realize the up/down/left right buttons had been
updated to send CC's on the latest firmware. I will download now.

Thank you,

closed 2021-09-14 by mat1jaczyyy