mat1jaczyyy/lpp-performance-cfw #63: Note on/off behaviour in chord mode when pressing two pads that contain the same note simultaneously

jaard, 2021-09-01:
First of all, thanks for this great project! I'm very interested in contributing if I can.
But let's first outline the issue I observed:

Intended behaviour (I assume):
If the user presses two chords that contain the same note and then one of them is released, the notes that are part of both chords remain on (i.e. no note off message is sent).

Current behaviour:
When one of the two chords is released, a note off command is sent for every note in that chord. The second chord that is still held down is missing notes.

Proposed solution:
I was building an Arduino-based Midi-controller some time ago and ran into the same issue. The way I solved it was by introducing a matrix of integers for every possible Midi note, initialised with zeros. When a button on the controller was pressed, that note would get a "note on"-message and the corresponding value in the matrix would get incremented by 1. When a button on the controller was released, the value in the matrix would get decremented by 1. But a "note off" message would only get sent if the value in the matrix was zero at this point. If its not, that means another chord is still holding that note.

Hope I made it clear! Would be grateful for suggestions where to start with this, what files or structure would need to be changed etc. Then I'm happy to contribute.

mat1jaczyyy, 2021-09-01:
Hey, thanks for the issue report!

The intended behavior is whatever the Pro MK3 does in their official Chord mode, I'll need to double check what Novation does there.

jaard, 2021-09-01:
Do you own a MK3 as well? I was eyeing it since it came out but just realised that you already implemented the MK3 chord mode, so that was great news.

mat1jaczyyy, 2021-09-02:
I don't actually, but I know some people who do.

mat1jaczyyy, 2021-09-02:
Your assumed intended behavior matches the Pro MK3, I'll see when I can catch some time to do fix this up, I don't think it will be too soon...

If you want to contribute and fix it yourself, `chord_press` in this block returns the pitches (of the chord) affected by the button that was just pressed. It currently has no memory of what other physical keys are pressed down, so it clears up pitches that might be held by other keys. You will probably need to use additional memory to achieve this, that is fine so long as it's efficient (a `u64` treated as a bitfield for keeping state of the 6x8 grid can be compared quickly and doesn't use much memory).

https://github.com/mat1jaczyyy/lpp-performance-cfw/blob/dfdc951f0a44b770fb52e573ccafd5a8b76119b2/src/modes/normal/chord.c#L396

mat1jaczyyy, 2021-09-02:
@jaard 
4

jaard, 2022-06-11:
Sorry @mat1jaczyyy, I completely missed your comment in September last year.
I just started using my Launchpad again, that's why I came to check for any news.

Thank you for pointing me to the fix, I'll have a look and try to contribute. I may have to ask you some more specific questions later.