mat1jaczyyy/lpp-performance-cfw #68: Ableton track color in Live Mode when pressing "Recording Arm"

AaronGrand, 2023-10-25:
Hey, this is a new Feature request.
It would be great to have Track color information in the stock Live mode, when pressing the "Record Arm".
To distinguish which track is where.

mat1jaczyyy, 2023-10-25:
I'm 90% sure that stock Live mode doesn't have this feature, going off memory here

For the built-in modes (Live, Note, Drum, etc) I generally want to stay as close to stock firmware as possible and not really add any massive enhancements as it gets out of control very quickly, especially as Ableton can technically change their control surface at will (I know they most likely won't).

Also I'm 99% sure Ableton doesn't send track color information to the firmware at all, so I'm not sure where I'd get the color info from without a modded control surface to pair up with.

I'll definitely double-check this at some point, but I don't have high hopes for getting it implemented.

mat1jaczyyy, 2023-12-16:
Double-checked, Ableton doesn't send track color information to the firmware

closed 2023-12-16 by mat1jaczyyy